debian-pi/raspbian-ua-netinst #524: PXE compatibility

jcea, 2020-07-31:
Raspberry PI 3B+ and 4B (and previous models with a microSD with a up to date firmware) can boot via PXE. That is, boot from network.

I have a minimal patch to this project to create files suitable for PXE boot, when booting a Raspberry PI 3B+.

You store those files in a PXE server, boot your raspberry pi with a blank microSD and the raspberry PI 3B+ will boot a (small) variant of this project from network and then do a regular raspbian install.

That is, you can deploy a raspberry with blank microSDs. You don't need to plug the microSD first to a computer and flash "raspbian-ua-netinst" to it.

Current state works perfectly fine and fullfills my needs. The changes, nevertheless, are so small and trivial that I think that this project would be interested. Current state is fine for me, but would need some work before a pull request. I would do that work if this project is interested in PXE booting.

Pending improvements:

- Be able of configuring installation using the regular configuration files. Not implemented yet because I don't need it, but easy to do.

- Support Raspberry Pi 4B. I will have access to my 4B in october. Somebody else could try before that time.

- Support older raspberry PI models. This would require a to use a preconfigured microSD with updated firmware, and in this situation you could just write on it regular "raspbian-ua-netinst". PXE boot could be useful for some usecases, maybe.

- For some reason I don't understand, I need to explicitly load the "lan78xx" driver (the Ethernet driver in Raspberry 3B+) in my patch in order to be able to enable networking. I don't know that is not needed when booting from the microSD card. Would be nice to investigate & resolve. I don't know the possible negative impact if trying to "modprobe" that module in a different raspberry PI model.

To sum up, would you be interested in expanding "raspbian-ua-netinst" to generate PXE files?

cseelye, 2020-09-21:
Are your patches available to look at/try out? I would like to do the exact scenario you are describing so that I can install/reinstall a group of Pis without touching the SD cards.

diederikdehaas, 2020-09-29:
It could potentially be interesting.
One problem is that I have no experience with PXE and I don't tend to include things I haven't tried myself.
Also, I don't have a RPi4 and I don't know if or when I will get one. I do have RPi3B+'s.

> Supporting older raspberry PI models. This would require to use a preconfigured microSD with updated firmware, and in this situation you could just write on it a regular "raspbian-ua-netinst". PXE boot could be useful for some usecases, maybe.

"preconfigured microSD with updated firmware" scares me a bit and may cause me to reject this idea. So you'd have to provide details that would/could convince me.
Even though this project isn't [Reproducable](https://reproducible-builds.org/), I do find that principle very important.