debian-pi/raspbian-ua-netinst #526: Failed to install minimal: "Operation not permitted" while creating initrd hard link

lukehutch, 2020-08-10:
See screenshot. The only thing I changed in the settings before booting was to set the distribution to "minimal". The install continued after this error, and the RPi appears to boot OK.

[image]

typhoon71, 2020-08-18:
I get the exactly same error every time I upgrade something, usually security updates.

diederikdehaas, 2020-08-18:
That's because FAT32, ie `/boot/` doesn't support hard links. So this is fully expected.
And indeed, every time there is f.e. a kernel upgrade and thus dpkg wants to create a hard-link (as backup), you'll see that warning/error.

closed 2020-08-18 by diederikdehaas

typhoon71, 2020-08-18:
Then a question arises: back on Jessie this didn't happen, did something change? 
It was fat32 back then too, maybe it wasn't trying to create hardlinks? Why are those create anyway?

diederikdehaas, 2020-08-19:
It's not related to Jessie, but a change in the installer over time. My guess it's specifically [this change](https://github.com/debian-pi/raspbian-ua-netinst/commit/a7e80f0dba793cd38945b596a9fd4b3b843d7bbb) and/or [this one](https://github.com/debian-pi/raspbian-ua-netinst/commit/93ad25602e6900530b9baabcd6c2e10aa824f22e), included in the installer since v1.1.1.

In the screenshot, you can see "**Re**generating initramfs", which is also triggered with f.e. a kernel upgrade. And with an upgrade, `dpkg` first wants to make a backup in case things fail.
It has always been FAT32 as the ARM GPU, which starts the RPi boot up process, can only read from that filesystem type, hence it has always been a requirement.