debian-pi/raspbian-ua-netinst #539: [Question] How are chances for Long-Term-Support / Updates?

pico-prob, 2023-07-05:
I have a headless rpi1 on an "open" colocation. There's no management tools, vpn, kvm, etc. So I am looking for a tool to handle OS-updates more safely. I came across this tool and it looks very suited (and straight forward) for my purpose. 

**However there haven't been any changes to this repo for 2 years. And now I am wondering, if you are planing on continuing on this installer?**
I am not talking about new features, just future compatibility. It would make sense for me that there's not much change in this sector (regarding my legacy hardware)..?

In any case thanks for this neat tool!

Cheers,
Mathias

chris18890, 2023-07-06:
hey @pico-prob, some of the work was forked off into https://github.com/FooDeas/raspberrypi-ua-netinst/, which may help you :)

pico-prob, 2023-07-06:
Thanks @chris18890 for the quick reply. FooDeas fork seems to bee quite sleepy as well. Last commit around the same time.
I ran across [pinn](https://github.com/procount/pinn). It's a bit more than I need though. Any advice on how to solve my problem?

closed 2023-07-06 by pico-prob